SpenceKonde/AVR-Guidance #1: PIN_Pxn not uniform

EclipseBoom, 2020-06-19:
Using the MegaTinyCore and PIN_Pxn references do not work for Serial.pins() or analogRead().  Under this scheme is pin A3 (digital pin 16) referred to as PA3_PIN for both digital and analog functions?

Just commenting because you recommend using this notation so I was attempting to "do it right" but got stymied pretty quickly.

SpenceKonde, 2020-06-19:
Thanks!

PIN_PA3 should work for all purposes. If it does not. please let me know the code you're using and what error or unexpected behavior you're seeing.

PA3_PIN should not work for any (where did that reversed spelling come from?)

In general, with this repo, do bear in mind that it's not ready for public consumption yet. I do very much want to start getting people on board with this new notation, though. The whole idea of "analog pins" and "digital pins" as if they were different (when its more analogous to certain pins can output PWM in addition to normal functions) causes a great deal of confusion, and then there's---- okay, I need to stop or we'll be here all day...

megaTinyCore uses the same pin numbers for analog and digital. ATTinyCore, probably has some issues here (I started using PIN_xn - then the official core started using PIN_An to mean something entirely different!)

EclipseBoom, 2020-06-20:
Thanks for the reply.  I just tried it and it does work if you use the proper syntax.  Imagine that!!!

I probably reversed them because I tend to make defines in my code with names like  "FUNCTION_PIN".  Something like "#define STATUS_LED_PIN    5"  in my code will translate to PIN_PB5 with this style.  I do like this style and wholly approve of it for all the reasons you state.  I can't even remember how many times I've looked up the pinmap to see what digital pin number is assigned to some "Analog" pin.

Once you get beyond blinky LEDs and the Mega328 you need to be able to refer to the hardware not some simplified invisible arbitrary pinmap.

closed 2020-06-20 by EclipseBoom