SpenceKonde/AVR-Guidance #5: possible solution for upload problems with CH340 Nano clones

1larity, 2021-06-30:
Not really an issue, but a problem and solution. I was having no joy uploading using a chinese (CH340) Nano clone to a 1614 F and found it would only work with no resistor in circuit with the UPDI pin.  Avrdude was reporting either no power or incorrect state errors during upload.
Hopefully this might be helpful to someone experiencing the similar issues in future.

closed 2021-06-30 by 1larity

SpenceKonde, 2021-06-30:
I think this is caused by old version of jtag2updi firmware, or that's what I thought was the issuer. It was one of the things that led me to stop recommending them aas programmers, as opposed to just using serial adapters.

1larity, 2021-07-01:
Unfortunately the serial programmers have thier own problems, especially on WIndows 10 where the manufacturers drivers fail installation (even in unsinged driver mode). I've brought and failed to get 5 different CH340/341 boards working for various different reasons. The Nano was my last ditch effort, and I think, much more accessible and available fro non-commercial users such as hobbiests and makers, than buying raw USB-serial chips and trying to build a programming device from that.

Anyway, thanks for the excellent work in making the new chips accesible for makers!

SpenceKonde, 2021-07-02:
Weird. I have literally never had a problem with CH340G serial devices on any of my systems. including half a dozen Win 10 ones, and I didn't have to turn off drivier signing or anything.

In any event, you must have rtesolved those problems if you say you're using a CH340G nano clone! If you can talk to a CH340G -based nano clone, your CH340 drivers are all set.

SpenceKonde, 2021-07-02:
Though I will note that I have encountered few DOA serial adapters from china, and some that died if you put them into 3.3v mode. And all the ones that come with a full-size USB plug on the  need the tabs on said connector actually soldered in place. (Idop this with USB hubs topo = each of those connectors has =2-6 tabs tjhat help hold it in place. Often cheap ones have one one of them soldered in place on each connector (and you can see trace of rosin flux around it, demonstrating that it was hand soldered.