SpenceKonde/AVR-Guidance #9: Collections has no attribue Hashable

dork3nergy, 2022-02-23:
Not sure if this is an issue on my side but, I get this error trying to upload.  

My python version is 3.10.2

---------------------------------------------------------------------------------------
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port /dev/ttyUSB0 at 460800 baud.
Target: attiny1614
Set fuses: ['2:0x01', '6:0x04', '8:0x00']
Action: write
File: /tmp/arduino_build_669110/Blink.ino.hex
Traceback (most recent call last):
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/prog.py", line 285, in <module>
    main()
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/prog.py", line 127, in main
An error occurred while uploading the sketch
    setup_logging(user_requested_level=logging_level)
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/pymcuprog/pymcuprog.py", line 47, in setup_logging
    configfile = yaml.safe_load(file)
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/__init__.py", line 94, in safe_load
    return load(stream, SafeLoader)
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/constructor.py", line 37, in get_single_data
    return self.construct_document(node)
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/constructor.py", line 46, in construct_document
    for dummy in generator:
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/constructor.py", line 398, in construct_yaml_map
    value = self.construct_mapping(node)
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/constructor.py", line 204, in construct_mapping
    return super().construct_mapping(node, deep=deep)
  File "/home/eturd/.arduino15/packages/megaTinyCore/hardware/megaavr/2.5.10/tools/libs/yaml/constructor.py", line 126, in construct_mapping
    if not isinstance(key, collections.Hashable):
AttributeError: module 'collections' has no attribute 'Hashable'

```

SpenceKonde, 2022-02-23:
This seems complicated, because we have to make sure that any change doesn't break compatibility for any of the platformws which this change doesn't occur on. Note that windows brings in it's own static installation of python, whereas linux and mac use the system installation. I don;t know how you're supposed to handle things like this - I'm a C programmer, not a python programmer.

dork3nergy, 2022-02-23:
I see your point.  You can pull in the version of python running your script and then have some if statements to deal with different version levels.

Although the collections.abc.hashable fixes the error for me, I am still unable to write to the chip. Since this is my first shot at this I'm not positive the problem isn't somewhere else. If you don't mind, I will send you an email via tindie (I have some of your dev boards) to elaborate.

SpenceKonde, 2023-07-05:
This issue has been fixed as far as I am aware in recent versions of SerialUPDI.

closed 2023-07-05 by SpenceKonde