SpenceKonde/AVR-Guidance #11: Simplify UPDI documentation

technoblogy, 2023-01-12:
The page:

https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md

is a bit daunting for someone just wanting to get UPDI to work with DxCore or megaTinyCore.

It's great to have all that information available, but first there should be a simple circuit that will work in most cases. After reading all your advice I think this should be:
````text
--------------------                                   To Target device
                DTR|    470Ω    BAT43                 __________________
                Rx |---------------------,-----------| UPDI
                Tx |----/\/\-----|<|----'       ,----| Gnd
                Vcc|---------------------------/-----| Vcc
                CTS|                          /      |__________________
                Gnd|-------------------------'
--------------------
````
I believe this will work whether or not the USB to Serial adapter has an internal resistor in series with Tx, or the target device has a resistor in series with UPDI. For most people the BAT43 will be easier to solder than an SMD diode, and it's available at most hobby shops.

Put this first, and then have all the suggestions about what to do if you don't have a Schottky diode available, etc., under a Further Information heading.

SpenceKonde, 2023-01-13:
That I realize that part could be better. I am so deep into that stiff that
I'm bad at explaining it sometimes. If you could send in a pr that combines
fixes and annotations where information is missing that would be great a d
would expedite the process as this plays to my weak points( even if it's
not complete) and then I can build on that

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2023-07-05:
Hmm, was that diagram on the original issue? 

Because I notice that you've moved the 470 ohm resistor to the other side of the diode. I am.... unsure I guess is the best word.... of what the ramifications of this would be. Can you describe the reasoning behind that proposed change? 

This change would put me in a really awkward business pickle. Continue to sell hardware that doesn't conform to my recommendations? Or take a loss of $1000-2000 to provide hardware that does, after a 1 month delay for modification and production of every modern AVR design I sell, which doesn't support that configuration. Being as I'm getting forced to relocate again at the end of august, which a couple grand on moving expenses alone and yet more for inevitable knockon expenses, and considering what tindie has brought in lately (not fucking enough) I do not have the resources for the second option.  So I'd have to sell boards with a known defect. But as I describe below, I think your approach has a very serious defect if you consider it in the context of designing a safe pcb for any available programmer.

There are other issues with your proposed schematic, beyond that it puts me in an awkward position

1. It implies that everyone needs to rewire their SerialUPDI devices
2. It requires use of 2 components, not 1 outside of what is normally integrated on the microcontroller board (or could be) (I put a resistor in series with the UPDI line on the PCB)
3. If a jtag2updi device assembled in the most casual way possible (direct wire from the IO pin to UPDI pin, almost ubiquitous in the field) a maximally desynced state could potentially cause damage to the hardware of one or both devices. 
  a. I don't want that to happen to my customers, as it would lead to time consuming support inquiries, poor reviews and requests for refunds. 
  b. That's why there's a 470 ohm resistor on the board - it's admittedly a 1-size-fits-some-better-than-others case, but on a Dx-series, that UPDI pin is not a weakass reset used-as-IO pin. It's a motherfuckin full fledged Dx-series I/O pin, likely with the usual full drive strength (tinyAVR has "limp noodle" pin drivers on A0.). With 470 ohm resistor, it is "safe" - if the programmer and target tried to drive the pin in opposite directions, it stays well below the maximum (I may lower the value of this resistor for the Dx/Ex-series parts - For Dx series on both sides (and you ain't gonna find anything with beefier pin drivers on normal pins) at the abs. maximum current of 50mA running from 5.5v (worst case scenario), V<sub>OH</sub> = 3.5V, V<sub>OL</sub> = 0.6V  (what a difference! .6V voltage drop sinking, and 1.5V drop while sourcing!) but anyway, that states that at 50mA = abs. max, we would need a resistor with a drop of 2.9V or more at 50 mA. a 100 ohm resistor would easily be sufficient, and I may start assembling the the Dx and Ex series parts like that. Unfortunately I don't know what the absolute max current through PA0 when used for GPIO or UPDI is.... (I'm actually going to ask this and a related question about Dx/Ex and HV UPDI of my contact on the inside) It may be than 100 ohms even is a better choice that purpose.  My goal is just to make sure I'm not allowing for naively constructed jtag2updi devices to damage either device's I/O pins. 
4. It would mean that I should place another order with the board house with a rev. b of my UPDI programmers (both of them) - Rev. A* still hasn't finished PCB production, because it means they're "wrong" now too. 

Can you share what your reasoning was behind proposing a different topology? How would you address the fact that is is impossible to make a board safe to use with a naively constructed jtag2updi (these are still extremely common in the wild, as many people are using that as SerialUPDI, and putting the resistor on the programmer side is rare) which is compatible with that wiring scheme?

* - The Rev. - was not even assembled because a defect was discovered on another board that used a "xerox" of the power supply switching mechanism, which neglects to account for the fact that MOSFETs have a body diode, which prevented it from correctly switching to a low voltage.

technoblogy, 2023-07-05:
The diagram I copied from your original document was the one I thought you were recommending, which was:
````text
No internal resistor
--------------------                                   To Target device
                DTR|                                  __________________
 No resistor    Rx |---------------------,--/\/\-----| UPDI---------------->
                Tx |--/\/\---|<|----\/\/'        .---| Gnd
 This is rare!  Vcc|---------------------------------| Vcc
                CTS|                          .`     |__________________
                Gnd|-------------------------'
--------------------

````
where you say that the resistor can be in ONE of the three positions. I just chose one of them, to make it simpler. It would be just as suitable to have the diode and resistor positions swapped. I was trying to find the one that will work in 99% of situations.

Perhaps I've misunderstood what the "best" circuit" should be, in which case by all means change it. I certainly didn't intend to cause you problems with your board sales.

My main point was that you have given people 10 alternative wiring diagrams, which is daunting/confusing. Give them one that will probably work, and then if they have a problem, or want to save components, they can then read the "Advanced suggestions" section later in the README.

PS Sorry to hear that you're being forced to relocate. I hate moving, so I can imagine how you feel.

technoblogy, 2023-07-05:
By the way - feel free to totally ignore my original comment. It was just a suggestion.